lsd-maddrive/STM32_ChibiOS #7: [HAL_topics/EXT_setup.md] Исправить инфу о каналах модуля EXT

serykhelena, 2018-11-27:
В даташите я нашла, что у EXT 16 каналов (начиная с 0-го), но при конфигурации этого модуля - там получается больше строк. Возможно, я заблуждаюсь в процессе выбора пина для генерации внешнего прерывания. Можешь посмотреть, пожалуйста (=

KaiL4eK, 2018-11-27:
Немножко обнаглею, но можно ссылку на страницу?) В метро еду)

serykhelena, 2018-11-27:
Прошу (= https://github.com/KaiL4eK/STM32_ChibiOS/blob/master/HAL_topics/EXT_setup.md

KaiL4eK, 2018-11-27:
В DS плз :neutral_face:

KaiL4eK, 2018-11-27:
Про 16 каналов

serykhelena, 2018-11-27:
https://www.st.com/content/ccc/resource/technical/document/datasheet/group3/c5/37/9c/1d/a6/09/4e/1a/DM00273119/files/DM00273119.pdf/jcr:content/translations/en.DM00273119.pdf#page=26 

На всякий случай, стр. 26

KaiL4eK, 2018-11-27:
Дело верное указано, смущает наличие 25 компараторов, но, думаю, правильно отмечено, что ног у портов обычно 16, так что не паримся и работаем только с 16ю элементами
Я ещё капну в исходники, но для действий по умолчанию лучше пользовать 16 элементов 🤔

KaiL4eK, 2018-11-27:
В исходниках вообще переменная EXT_MAX_CHANNELS установлена в 20..
Тему закрываю, используем 16 каналов

closed 2018-11-27 by KaiL4eK