lsd-maddrive/STM32_ChibiOS #22: [Linux_setup] Уточнить

serykhelena, 2019-02-19:
Не получается запутить скрипт load.sh из терминала. 
Нагуглила, что нужно прописать ./load.sh - говорит, что нет разрешения.
Можешь, пожвлуйста, поподробнее написать, что нужно сделать.
Правильно я понимаю, что, например, если я хочу писать код просто в текстовом редакторе (например, Sublime Text), то из терминала я могу собирать проект и загружать?

KaiL4eK, 2019-02-19:
Добавил инфо, есть подозрение, что со скрипта слетаю права, если репозиторий закоммитить из винды. В этом плане инфа должна быть универсальна, так как решать такую проблему на лялихе (Linux) приходится часто.

KaiL4eK, 2019-02-19:
А и ответ на вопрос, да, можешь, конечно) В инструкции это даже где-то есть.

closed 2019-02-19 by KaiL4eK