lsd-maddrive/STM32_ChibiOS #23: [Linux_setup] Вопрос про отладчик

serykhelena, 2019-02-19:
Важно ли в какую папку клонируешь репозиторий `openocd`?

KaiL4eK, 2019-02-19:
Сама по себе концепция сборки предполагает, что исходники могут валяться где угодно. Важно укахать точный путь инсталяции. Для целей укомплектования всех собираемых либ в инструкции создается папку `ext_repos`, чтобы не мусорить в домашней папке и при желании посе инсталяции снести одним жестом.

KaiL4eK, 2019-02-19:
Нужно добавить инфу о том, куда клонировать?

serykhelena, 2019-02-19:
Мне кажется, лучше добавить (=

closed 2019-02-19 by KaiL4eK

serykhelena, 2019-02-19:
Спасибо! 
А скажи, что ты имел в виду "Переходим в папку cd openocd и делаем сборку по readme:"? 
Нужно открыть readme и сделать Quick Start или всё же использовать команды, которые ты ниже приводишь? Мне не очень понятно )=

KaiL4eK, 2019-02-19:
Видать лишнего написал, думаешь стоит убрать? В README просто описано, как собирать, шаги абсолютно идентичные)

serykhelena, 2019-02-19:
Может так и написать, что в README можно посмотреть более подробно, что и как собирать, но команды ты уже и так приводишь? (=