lsd-maddrive/STM32_ChibiOS #25: [Linux_setup] PATH и Eclipse

serykhelena, 2019-02-19:
Может быть немного уточнить инфу про то, что нужно прописать переменную PATH.
Это достаточно сделать в терминале или нужно также добавить в .bashrc ?

KaiL4eK, 2019-02-19:
Хорошее уточнение, сами переменные предполагают установку на нынешний терминал (сессию), но если вкатить в *rc файл и вспомнить, что этот *rc файл запускается при каждом создании, то получаем установку на все терминалы.
Инфу добавил, достаточно ли этого или нужно еще конкретизировать?

serykhelena, 2019-02-19:
Добавила в .bashrc, но не помогло. 
В терминале пробовала вводить просто ChibiStudio и ./ChibiStudio, автодополнение не работает, пишет команда не найдена. Я что-то не так делаю?

KaiL4eK, 2019-02-19:
Где лежит папка ChibiStudio?

serykhelena, 2019-02-19:
в Downloads

KaiL4eK, 2019-02-19:
А PATH, полагаю, прописан до `$HOME/ChibiStudio`?

serykhelena, 2019-02-19:
Да, до меня теперь тоже дошло... 
Сейчас поменяю (=

KaiL4eK, 2019-02-19:
Я бы указал на это в инструкции, но пускай это будем маленькой заковыркой, чтобы размяться =)

KaiL4eK, 2019-02-19:
А то мы тут о cmake, configure, сборке пакетов говорим, так вот пусть будет маленький сюрпрайз)

serykhelena, 2019-02-19:
Хо-хо-хо! (=
Теперь всё работает (= Закрываю issue

closed 2019-02-19 by serykhelena

KaiL4eK, 2019-02-19:
В инструкции пишутся фразы "Перейдя в папку" и "указав переменную PATH до этой папки", так что пускай разбираются)