lsd-maddrive/STM32_ChibiOS #26: [hal_topics/setup -> EXT] Уточнить про структуру конфигурации

KaiL4eK, 2019-03-10:
Долгими мучениями дошли до того, что при вызове `extStart()` конфигурационная структура должна быть статической или глобальной. Драйвер сохраняет указатель на структуру вместо самой структуры. Работа с локальной структурой (выделена на стеке) может привести к тому, что память, на которую ссылается сохраненный адрес, будет переписана и дальше `undefined behaviour`.

RuslanZhuch, 2019-04-08:
Аналогичная ситуация с icu
Моментально выбрасывает в цикл для необработанных исключений при вызове коллбэков

KaiL4eK, 2019-06-15:
Так как на новой версии модуль удален - вопрос исчерпан. Проблемы с локальным выделением структуры считаю стоит решать локально. Тем не менее бага будет в истории - будем знать)

closed 2019-06-15 by KaiL4eK